Traumflug/Teacup_Firmware #304: Double Z-Axis motors

thomaskilian, 2018-09-25:
I wonder if it's possible to configure the FW for 2 Z steppers?

phord, 2018-09-25:
It's usually easier to wire the signals commonly between the two Z-stepper drivers, or to chain the two steppers (in serial) on the output of the one Z-stepper driver.

phord, 2018-09-25:
That said, it wouldn't be hard to support two Z-stepper outputs in the code.  Look for `z_step` and `z_direction` in the code.  You could probably add all the needed changes in the macro definitions in `pinio.h`.

thomaskilian, 2018-09-25:
Thanks. I'll give that a try. Actually I'm moving away from Marlin since the FW starts behaving weird and it's almost impossible to maintain. I got two MAX6675 which seemingly can be incorporated more easily with Teacup. But the 2 Z is probably more tricky since I also have 2 end-stops for homing.