Traumflug/Teacup_Firmware #307: Thermistor table generated wrongly

thomaskilian, 2018-10-04:
I can't run the configtool.py on my Mac (no input possible; I don't want to go through that Python version hell to find the wx-issue). Anyhow, I tried to generate the temptable with createTemperatureLookup.py but failed. It creates a 2-dim array like

    const uint16_t temptable[NUMTEMPS][2] PROGMEM = { 
    // {ADC, temp*4 }, // temp         Rtherm     Vtherm      resolution   power
       {   1,   2433}, //  608.44 C,        2 Ohm, 0.005 V, 118.68 C/count, 0.02mW

etc. So the access fails when trying to access the 3d dimension:

    /Users/thomaskilian/Documents/Teacup_Firmware/temp.c: In function 'temp_table_lookup':
    /Users/thomaskilian/Documents/Teacup_Firmware/temp.c:175:48: error: subscripted value is neither array nor pointer nor vector
         if (pgm_read_word(&(temptable[table_num][j][0])) >= temp)
                                                    ^

Any quick fix for that?

triffid, 2018-10-04:
should be

```
temptable[NUM_THERMISTORS][NUMTEMPS][2] PROGMEM = {
    { // heater 0
        {  1, 2433},
        { ...
        ....
    },
    { // heater 1
        { ...
       ...
    }
    ...
}
```

I guess since we support multiple temperature inputs, and note you may need to multiply the temperatures by 4 since last I checked, teacup still uses 14.2 fixed point for temperatures.

thomaskilian, 2018-10-04:
Seems like that fixed the syntax. I'll see what temps it will spit out now :-) Thanks!

Wurstnase, 2018-10-18:
Looks like we need to rework the createTemperatureLookup.py.

Some time ago @phord reworked the part in the config tool. So the table becomes much more accurate for all values.

AnHardt, 2018-10-19:
Don't know how the the search for values is done in teacup but maybe it's worth to minimise the amount of values. One could search the turning point and search from there to the ends, for lines not more apart from the ideale funktion than a defined error.