Traumflug/Teacup_Firmware #310: How to change the serial port for melzi board

Laczen, 2018-10-22:
Hi, I am trying to modify Teacup to serial port 1 instead of 0 on a melzi board (1284p). If I modify the pin numbers for RXD and TXD nothing changes. In the serial code I can also not find any references to these pins. How can I change the configuration so that serial port 1 is used ?

Thanks

Wurstnase, 2018-10-22:
The serial for AVR is actually hard coded. But didn't use the melzi the serial 0 with the USB port?

Laczen, 2018-10-22:
Hi @Wurstnase, thanks for the reply. Serial 0 is connected to USB and is used also used for flash updates. I would like to connect the melzi board to a rpi where the USB ports are all used. I have been looking at serial-avr and could change all references to uart0, but also uart1 seems to be used (intercom.c). I am not sure where I would need to make the changes to get everything working.

Wurstnase, 2018-10-23:
I'm pretty sure, that you don't use the intercom board. This is a very special and today uncommon thing. Intercom is a board, where you split the extruder and the movement. The Melzi and 99.9% of all other board will do both jobs.

You should keep care connecting the board directly to the RPi. The RPi is only 3.3V compatible and the Melzi will send 5V. You need at least a voltage devider for TX (Melzi) to RX (RPi).

Traumflug, 2018-10-23:
Serial is set up here: https://github.com/Traumflug/Teacup_Firmware/blob/master/serial-avr.c#L74-L88

Laczen, 2018-10-23:
Hi @Wurstnase and @Traumflug, I have been able to change the serial port from 0 to 1 by making changes in serial-avr.c. Thanks for the provided help.

Wurstnase, 2018-10-24:
Can you please share your changes? So others could find it.

Laczen, 2018-10-24:
Hi @Wurstnase, I have no problem with sharing. I wouldn't put it in the github tree because it might be confusing for others. 
[serial-avr.zip](https://github.com/Traumflug/Teacup_Firmware/files/2510818/serial-avr.zip)