Traumflug/Teacup_Firmware #311: Any pointers on how to attach a servo?

thomaskilian, 2018-11-04:
I wonder if there are any pointers on how to attach a servo? I know that M280 is not supported (because it's a > 255 code) but would not mind misusing anything here.  Basically I need to send just two servo positions (aka two PWM codes but with the servo PWM freq. of 50 Hz).

phord, 2018-11-06:
Make sure to choose a PWM pin that uses Timer0.  The other pins are probably not configured correctly, and Timer1 is used internally.

thomaskilian, 2018-11-18:
@phord I'm a bit lost now. I tried a simple sketch (on a Mega2560) which produces nice servo ticks. But once I implant that (adapted) code into Teacup it loop-resets permanently once I start timer3. Any idea?

    void setup() {
      // put your setup code here, to run once:
      for (int i = 3; i <= 13; i++) {
        pinMode(i, OUTPUT);
        digitalWrite(i, 1);
      }
      digitalWrite(3, 0);
      digitalWrite(9, 0);

      TCCR3A = 0;
      TCCR3B = 0;
      TCNT3  = 0;
      OCR3A = 199;
      TCCR3A |= (1 << WGM31);
      TCCR3B |= (1 << CS31);
    }

    int servoTicks;
    int servoWidth;
    int servoDuty;
    void timer0_start(int i) {
      cli();
      servoTicks = 0;
      servoWidth = i;
      servoDuty = 10; // 10 ticks ~ 0.2 sec
      digitalWrite(3, 1);
      TIMSK3 |= (1 << OCIE3A);
      sei();
    }

    ISR(TIMER3_COMPA_vect){
      servoTicks++;
      if (servoTicks >= 80) {
        servoTicks = 0;
        digitalWrite(3, 1);
        servoDuty--;
      } else if (servoTicks == servoWidth) {
        t[ptr++] = micros();
        digitalWrite(3, 0);
        if (servoDuty <= 0)
          TIMSK3 &= ~ (1 << OCIE3A);
      }
    }

    int w = 10;
    void loop() {
      delay(2000);
      cli();
      timer0_start(w);
      w = 15-w;
      sei();
    }

thomaskilian, 2018-11-24:
Never mind.It seems as if I need to enclose `TIMSK3 |= (1 << OCIE3A);` in `cli/sei`. 

Not working still, but the reset loop is gone.

thomaskilian, 2018-11-24:
I'd really appreciate some help now. I rather figured out how the heating PWM works. I'm using DIO7, 9  and 10 for bed, extruder and fan. On my Mega this will involve timers 2 and 4 for the PWM. So I used timer3 for my own purpose. I did set it like in the code snippet I posted above (without the ISR just resetting TIMSK3 once called). But after 5 seconds or so the Arduino gets a reset. Where is it that timer3 interferes somewhere the Arduino is reset???

Wurstnase, 2018-11-25:
First I would do very simple thing with the ISR for testing. Probably you could send one single char (ISR should always do its stuff very fast) through the serial line.

thomaskilian, 2018-11-25:
I don't know what would help me in that. I'm after a way to create a 50Hz signal with varying duty lengths (of about 1ms).

thomaskilian, 2018-11-25:
Yes. The timer is definitely triggered, but after some 5 seconds the whole thing resets. I implemented it in heater-avr.c in the `#ifdef	TCCR3A` part and below the timer start routine/ISR

Wurstnase, 2018-11-25:
Please share your complete project/code somewhere.

thomaskilian, 2018-11-25:
[Dropbox](https://www.dropbox.com/s/iwogjcqrbxnoorf/Teacup_Firmware.zip?dl=0)

I removed stuff I did not need. heater-avr.c and gcode_process.c contain the most important modifications. It's just hardcoded for the Atmega use (port 3 shall receive the servo PWM).

Wurstnase, 2018-11-25:
How did you test your changes? I see that you've implement a test gcode for yourself.

thomaskilian, 2018-11-25:
Yup. I just send a "G2", "G2 P0" or "G2 P1". They all trigger the timer once. For my purpose that will done since I anyway write the SW for the servo movement myself (rather than using the unreachable position servo code somewhere above 255).