Traumflug/Teacup_Firmware #312: Atmega128 Teacup

shakenov-chinga, 2018-11-05:
Hello everybody!
I've designed a new board using Atmega128 (no real reasons, just had several of them). I took sanguinololu as an example: everything is almost the same, except pin numbers and heatbed and extruder heater are connected to digital IO, not PWM. Now I am trying to configure and build Teacup using configtool.
What I've done so far is I created arduino_128.h file, where I use [MegaCore](https://github.com/MCUdude/MegaCore) pin numbers, and created board.myprinter.h file, which is a copy-paste with little necessery changes from sanguinololu board.

Now, the problem is that in arduino_128.h file I can't understand what and how do I assign DIOn_PWM and DIOn_TCCR values for pins. I have a little idea for DIOn_PWM values assignment, from datasheet I got:
`
#define OC0	DIO12
#define OC1A	DIO13
#define OC1B	DIO14
#define OC1C	DIO15
#define OC2	DIO15
#define OC3A	DIO3
#define OC3B	DIO4
#define OC3C	DIO5
`
and these values are used in it, but I have no clue what to do with DIOn_TCCR.

Could you please help to understand this of to find some useful source about it.
Thanks in advance.

triffid, 2018-11-06:
`#define DIO14_TCCR TCCR1A` should work if OC1B is on DIO14.

DIOx_TCCR just a pointer to the timer configuration register so we can determine the appropriate timer to configure when you set the associated pin as a PWM (heater) output.

shakenov-chinga, 2018-11-06:
As I can deduce these are values we use for PWM pins. Can someone confirm them, please? 

OC0		DIO12	-			-
OC1A	DIO13	TCCR1A		COM1A1
OC1B	DIO14	TCCR1A		COM1B1
OC1C	DIO15	TCCR1A		COM1C1
OC2		DIO15	-			-
OC3A	DIO3	TCCR3A		COM3A1
OC3B	DIO4	TCCR3A		COM3B1
OC3C	DIO5	TCCR3A		COM3C1

Tried to build using configtool, but got this error:
`analog-avr.c:42:3: error: ‘DIDR0’ undeclared (first use in this function)`
`  DIDR0 = analog_mask & 0xFF;`
Where its decloration should be?

shakenov-chinga, 2018-11-06:
Hi everyone,
Seems like atmega128 as well as atmega16 (and maybe other atmegas) doesn't need DIDR0 register to use ADC, so I just added `#ifdef DIDR0` macro in analog-avr.c. I hope this will work.
Also I had little problems with timer.c, because compiler complains that TIMSK1 didn't exist, so I had to use macros again to use TIMSK register instead, like:
`	#ifdef TIMSK`
`		TIMSK = 0;`
`	#else`
`		TIMSK1 = 0;`
`	#endif`
Now I have problems with heater-avr.c file, because it uses, as I can understand, timer/counter with pwm little differently. Can anyone tell, please, what is the logic in `heater_init()` function in this file and why so many TCCR registers are used? Is it there to support several possible extruders?

thomaskilian, 2018-11-24:
I'm struggling with a similar issue. I partly figured it out. The timers are preset to run at the desired PWM (fast) speed. Down below in the part

      if (PWM_TYPE(pwm, pin) == HARDWARE_PWM) {               \

it will set the appropriate timer registers to take care of the PWM.

Wurstnase, 2018-11-25:
Doesn't looks like the 128 has a lot of pwm-able pins. I would define all
without pwm, take a look into e.g. arduino1280 file, and use the software
pwm.

In the first 2 lines are two macros. Copy it in your file and name your
pins like the non pwm pins in the 1280.

If you need some help you could post your arduino file.