Traumflug/Teacup_Firmware #314: Can't save printer with Configtool

shakenov-chinga, 2018-11-06:
Hi.
I'm trying to save loaded and edited printer configurations. This error message is printed in terminal whenever I save, no matter if it is editted or not.

`Traceback (most recent call last):`
`  File "/home/chinga/Dropbox/Teacup_Firmware-master/configtool/gui.py", line 246, in onSavePrinterConfig`
`    rc = self.pgPrinter.onSaveConfig(evt)`
`  File "/home/chinga/Dropbox/Teacup_Firmware-master/configtool/printerpanel.py", line 198, in onSaveConfig`
`    return self.saveConfigFile(path)`
`  File "/home/chinga/Dropbox/Teacup_Firmware-master/configtool/printerpanel.py", line 246, in saveConfigFile`
`    v1 = pg.getValues()`
`  File "/home/chinga/Dropbox/Teacup_Firmware-master/configtool/mechanicalpage.py", line 242, in getValues`
`    result['HOMING_STEP4'] = self.getHomingValue('HOMING_STEP4', result)`
`  File "/home/chinga/Dropbox/Teacup_Firmware-master/configtool/mechanicalpage.py", line 231, in getHomingValue`
`    return (self.labels.keys()[self.labels.values().index(result[name][0])], True)
ValueError: 'n' is not in list`

Wurstnase, 2018-12-03:
I can confirm this issue. Looks like the button with '-' will return something different than expected. Also for other homing-keys if they are '-'. E.g. for Step 4 it is 'n'. For Step 2 it is 'y'.

Wurstnase, 2019-01-24:
Uff... Lot's of work.
Please check https://github.com/Traumflug/Teacup_Firmware/tree/configtool_homing
Feedback is welcome.

Homing is fixed. Also this should work with Python3 now.

I'm currently not very happy, with the double defines of homingKeys in [mechanicalpage.py](https://github.com/Traumflug/Teacup_Firmware/blob/a86d69ee3caf1a981c51c3ca7f0351befcf35f08/configtool/mechanicalpage.py#L44-L49) and [printer.py](https://github.com/Traumflug/Teacup_Firmware/blob/a86d69ee3caf1a981c51c3ca7f0351befcf35f08/configtool/printer.py#L34-L39). But I don't have any good idea for this.

sv1, 2019-04-06:
With python 2.7.13 on Debian, I can not run configtool at all.

$ ./configtool.py 
Traceback (most recent call last):
 File "./configtool.py", line 191, in <module>
    StartGui(getSettings())
  File "./configtool.py", line 38, in getSettings
    settings = Settings(None, cmdFolder, arg)
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 53, in __init__
    self.loaded = self.readConfig(ini)
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 78, in readConfig
    self.cflags = cfg_get("cflags")
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 73, in cfg_get
    val = cfg.get(key, values[key])
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 1302, in get
    fallback=fallback)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 816, in get
    d)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 401, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 450, in _interpolate_some
    u"found: %r" % (rest,))
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: u'%F_CPU% -mmcu=%CPU% -Wall -Wstrict-prototypes -std=gnu99 -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-sections -finline-functions-called-once -mcall-prologues -Wa,-adhlns=%ALNAME% -save-temps=obj'

Wurstnase, 2019-04-06:
I've currently not much time to do any support but probably this issue was already fixed in
https://github.com/Traumflug/Teacup_Firmware/tree/python_upgrade .

Can you please check?

sv1, 2019-04-06:
On Sat, 2019-04-06 at 11:12 -0700, Nico Tonnhofer wrote:
> I've currently not much time to do any support but probably this
> issue was already fixed in
> https://github.com/Traumflug/Teacup_Firmware/tree/python_upgrade .
> Can you please check?
> 

Same behavior ....

$ git checkout python_upgrade
Branch python_upgrade set up to track remote branch python_upgrade from
origin.
Switched to a new branch 'python_upgrade'
$ ./configtool.py 
Traceback (most recent call last):
  File "./configtool.py", line 191, in <module>
    StartGui(getSettings())
  File "./configtool.py", line 38, in getSettings
    settings = Settings(None, cmdFolder, arg)
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 53, in
__init__
    self.loaded = self.readConfig(ini)
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 78, in
readConfig
    self.cflags = cfg_get("cflags")
  File "/opt/linux/Teacup_Firmware/configtool/settings.py", line 73, in
cfg_get
    val = cfg.get(key, values[key])
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 1302,
in get
    fallback=fallback)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 816, in
get
    d)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 401, in
before_get
    self._interpolate_some(parser, option, L, value, section, defaults,
1)
  File "/usr/lib/python2.7/dist-packages/configparser.py", line 450, in
_interpolate_some
    u"found: %r" % (rest,))
configparser.InterpolationSyntaxError: '%' must be followed by '%' or
'(', found: u'%F_CPU% -mmcu=%CPU% -Wall -Wstrict-prototypes -std=gnu99
-funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums
-Winline -fno-move-loop-invariants -fno-tree-scev-cprop -Os -ffunction-
sections -finline-functions-called-once -mcall-prologues -Wa,-
adhlns=%ALNAME% -save-t

sv1, 2019-04-06:
On Sat, 2019-04-06 at 11:12 -0700, Nico Tonnhofer wrote:
> I've currently not much time to do any support but probably this
> issue was already fixed in
> https://github.com/Traumflug/Teacup_Firmware/tree/python_upgrade .
> Can you please check?
> —
> 

Started with a new/clean repo and now I have this:

$ git checkout configtool_homing
Branch configtool_homing set up to track remote branch
configtool_homing from origin.
Switched to a new branch 'configtool_homing'
$ ./configtool.py 
Traceback (most recent call last):
  File "./configtool.py", line 191, in <module>
    StartGui(getSettings())
  File "/opt/linux/Teacup_Firmware/configtool/gui.py", line 695, in
StartGui
    frame = ConfigFrame(settings)
  File "/opt/linux/Teacup_Firmware/configtool/gui.py", line 77, in
__init__
    self.pgPrinter = PrinterPanel(self, self.nb, self.settings)
  File "/opt/linux/Teacup_Firmware/configtool/printerpanel.py", line
54, in __init__
    self.pgMech = self.registerPage(MechanicalPage, "Mechanical")
  File "/opt/linux/Teacup_Firmware/configtool/printerpanel.py", line
65, in registerPage
    self, self.nb, len(self.pages), *args, font=self.settings.font,
**kwargs
  File "/opt/linux/Teacup_Firmware/configtool/mechanicalpage.py", line
175, in __init__
    b.SetToolTip("Open the calculator for axes that are belt-driven.")
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line
11390, in SetToolTip
    return _core_.Window_SetToolTip(*args, **kwargs)
TypeError: in method 'Window_SetToolTip', expected argument 2 of type
'wxToolTip *'

Wurstnase, 2019-04-07:
Please install the latest wxpython: https://wxpython.org/pages/downloads/index.html
3.0 is not supported with python_upgrade-branch.

Wurstnase, 2019-04-18:
Everything now in the experimental branch.