Traumflug/Teacup_Firmware #315: PID loop not running at full power

thomaskilian, 2018-11-13:
When I watched the signal LEDs for the bed/extruder I found that they are not working at maximum power, but only at 50%. My board runs at 24V and when measuring with a volt meter it reads 12V only (I could pul in my scope but for sure it will just show a 50% duty cycle for the PWM). Why is that so? 

My heater settings are 

`DEFINE_HEATER(extruder, DIO10,   1,      1,       100)
DEFINE_HEATER(bed,      DIO7,    1,      1,       100)
DEFINE_HEATER(fan,      DIO9,    0,      1,       50)`

Also, for the fan I noticed that it's not completely turned off but the LED is glowing very dimly. It's harder to measure but clear to see and, more important, clearly to hear by the fan making quite some noise in stand by.

thomaskilian, 2018-11-25:
I'll take one. Will post it later then.

thomaskilian, 2018-11-25:
Set to 5us per cm which means PW = 20us 

[image]

thomaskilian, 2018-11-25:
It "might" be, but why do I see that frequency once I start heating and I don't see it at the extruder? Both are configured with the `DEFINE_HEATER` macro (see top). I'll cross check the cables, though.

thomaskilian, 2018-11-25:
Both run a MOSFET in the same way. It's a MKS GEN V1. I'll try switching the outputs just to see whether the behavior will switch with the attached heat sources.

thomaskilian, 2018-11-25:
Never mind, that **was** an electrical issue. The PWM sends a short off (inverted logic) signal every 20us. The bed draws so much current that the power supply shortly breaks down so the strange pulse appears on the scope. Well, good to know at least :-)

You can close this.

Wurstnase, 2018-11-26:
You can close issue by yourself :)

closed 2018-11-26 by Wurstnase

thomaskilian, 2018-11-26:
Obviously I'm not much used to that. Will do next time.