Traumflug/Teacup_Firmware #318: ZProbing implementation

trustfm, 2019-01-09:
I am using teacup as a pick and place machine. 
I have variable height (Z) colums so i use ZMax endstop as sensor. 
I use the 
G162 Z100 Command for example 
This goes down 100 if i get no click i know that the columd is empty else i do a pick and place movement

This is implemented like this : 

Clock.h 
uint8_t ZPolling; 
int32_t ZMaxRuntime;

--------------
clock.cpp 

static void clock_10ms(void) {
	// reset watchdog
	wd_reset();

	temp_sensor_tick();

	ifclock(clock_flag_250ms) {
		clock_250ms();
	}
  
  
	//Get current Z position : by trustfm
  if (ZPolling==1) {
  	update_current_position();
    if (current_position.axis[Z]<(int32_t)(Z_MAX * 1000.) && current_position.axis[Z]>(int32_t)(Z_MIN * 1000.))  {
      sersendf_P(PSTR("ZCur: %lq\n"),current_position.axis[Z] );
      if (current_position.axis[Z]>ZMaxRuntime) {  
      
        sersendf_P(PSTR("noClick\n"));
		
        timer_stop();
        queue_flush();
        timer_init();
		
      }
    }
  }
                 
} 

----------------------------
gcode_process.c

			case 162:
				//? --- G162: Home positive ---
				//?
				//? Find the maximum limit of the specified axes by searching for the limit switch.
				//?
        #if defined X_MAX_PIN
          if (next_target.seen_X)
            home_x_positive();
        #endif
        #if defined Y_MAX_PIN
          if (next_target.seen_Y)
            home_y_positive();
        #endif
        #if defined Z_MAX_PIN
          if (next_target.seen_Z) {
            ZMaxRuntime=next_target.target.axis[Z];
            ZPolling=1; //by TrustFm
            home_z_positive();
            ZPolling=0; //by TrustFm
          }
        #endif
				break;

Now the question :  extremely rarely we get the machine stalled is there any possibility that specially 
        timer_stop();
        queue_flush();
        timer_init();
is the problem ? 
Can this be implemented differently ? 
Via serial i get "E:start" seems to restart arduino ...

I have tried to not flush the hole queue i have made a custom queue_next but does not seem to work as expected. 
Thank you for your time

Wurstnase, 2019-01-10:
timer_stop() is like riding a bike and throwing a stick of wood in your front wheel to stop.

Take a look into dda.c and the homing procedures. You could extend this with a new command, where the max end position is a bit different and it is not zeroing.

phord, 2019-01-10:
The homing works by setting the position to an extreme position (`dir * MAX_DELTA_UM`) and stopping when the endstop is triggered.  If you add a home command which uses a different target number instead of the extreme position, then I think it will mostly do what you want.

phord, 2019-01-10:
I have a commit that records the position where the endstop was triggered. I needed it for Z-Probing the bed using the BLTouch probe.  Maybe you can use something similar for your purposes.  The change is [here](https://github.com/phord/Teacup_Firmware/commit/5e2c3b0bd65a229f2551401e391fda6262847798), but I haven't tested it with many different printer configs.

trustfm, 2019-01-11:
Thank all of you for your support but my problem was bigger. 
Even with a fresh teacup i got stalls and random "start"strings using putty. 
For example 
(
M106 S255
G90
M114
M106 S0
G162
G28 Z
M114
G162 Z
M106 S255
G28 Z
M114
)

got errors and corrupted status messages like X:120.000start ...

I figured out that had to do with the XON/XOFF implementation and my crappy CH340 clone arduino board. I have tested another one with the FTDI chip (arduino duemillanove) and worked perfectly. 
I have found this source : https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl
"Streaming Protocol: Via Flow Control (XON/XOFF)"

What system hw and sw is suggested using teacup ? At this point the options are only FTDI chips ?

Wurstnase, 2019-01-11:
> got errors and corrupted status messages like X:120.000start ...

It could help, when you post exactly what you've send _and_ received.

Do you have any issues when XON/XOFF is disabled? XON/XOFF disabled is the default setting.

trustfm, 2019-01-11:
Hello i have some good news !
I have made a custom arduino trasnsmission tester CH340.
I have noticed that arduino clones using CH340 does not work (program stalls) but with trustduino (a mine single PCB arduino board) with the CH340 chip works !!!
I include the schematics and PCB of trustdino. The transmission tester is also in the zip file   
 
http://trustfm.net/hardware/Images/Trustduino/tinyserialv1.0.zip

PS : my crude endstop implementation with the 
timer_stop();
queue_flush();
timer_init();
sequence seems to work too with the FTDI and mine trustduino CH340 based board !!!