Traumflug/Teacup_Firmware #320: Compile error on Mint with Teacup Master Branch for ARM LCP1114

SanShell, 2019-09-06:
Good evening!

I'm trying to compile the firmware with the master branch for the ARM LCP1114.
I have followed the wiki about the ARM toolchain setup, the config.h is:
 #include "config/board.gen7-arm.h"
#include "config/printer.corexy.h" ( a mendel base tuned to COREXY)

Since the configtool is broken, the parameters were set by hands in the printer.corexy.h, the only thing that i have disabled is the watchdog (not implemented)

I receive this error:

~/Teacup_Firmware$ sudo make
  LINK      build/teacup.elf
lto1: fatal error: bytecode stream in file 'build/analog-stm32.o' generated with LTO version 5.2 instead of the expected 6.0
compilation terminated.
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile-common:93: recipe for target 'build/teacup.elf' failed
make: *** [build/teacup.elf] Error 1

Just to be sure, i have prepared a VM and a fresh install of MINT Linux. There i can compile and upload without issue, but the serial port is messed up. I can establish a connection fiddling around with the reset PIN, but on the serial port i receive START and nothing else.

I'm guessing about something broken with the compiler, right? If so, which compiler version i need to use?  Maybe in the meantime i try on the windows machine.

Thanks for any help

SanShell, 2019-09-06:
Finally, after many tries i did it, was the damn compiler.

I have dowloaded the arm-none-eabi version in the makefile (5_4-2016q3).

Now everything is working as it should.

closed 2019-09-06 by SanShell

Wurstnase, 2019-09-11:
Looks like in your build directory were some files already. Make will check and rebuild only files which need to build again. When you build before some files already and change then the compiler you simply need to ```make clean``` before a ```make```.

SanShell, 2019-09-11:
Forgot to mention, that the broken binary was compiled and uploaded without issue. The serial port was responding for a short while after a cold restart, I have no idea what the CPU was doing. My guess is that something has changed in GCC o mbed and created a conflict somewhere, but due to the lack of my own experience on programming with microcontroller, I wasn't able to create a solution and update the code.

Wurstnase, 2019-09-14:
I can read now 2 issues. 
First is two different compiler versions mixed.

The second should be fixed in the experimental. This is indeed a compiler bug with GCC 7.2.1 as far as I remember. With this bug, the weak symbols are not handled correctly in assembly code and can overwrite the non weak symbols.
In the experimental branch I fixed this by linking the startup code before the rest.
https://github.com/Traumflug/Teacup_Firmware/commit/0d308c8cfa90c7d1da8781fa25fc5e6ee69aed84