Traumflug/Teacup_Firmware #322: Cant Read NTC temp

azant, 2020-03-02:
i cant get temp reading on my HotEnd. i tried all different NTC presets and custom it never changes from 0.

so i downloaded a simple analog read and i can read temp. and also used Arduino NTC example and it worked. not sure if its something in my config. but id think i would get at least false reading instead of a Zero!

used Pronterface, manual Gcode commands.
Hardware: Arduino Pro mini
5v/16Hz

VVG-Tonnhofer, 2020-03-02:
Please share your config and maybe also the simple Arduino script. Else we can't help you.

Wurstnase, 2020-03-02:
Please share your config and maybe also the simple Arduino script. Else we can't help you.

azant, 2020-03-02:
attached Configs files. Thanks
[Configs.zip](https://github.com/Traumflug/Teacup_Firmware/files/4277068/Configs.zip)

Wurstnase, 2020-03-03:
Can you please also upload your thermistor table?

azant, 2020-03-03:
// default thermistor lookup table

// How many thermistor tables we have
#define NUMTABLES 1

#define THERMISTOR_EXTRUDER	0
// #define THERMISTOR_BED		1

// Thermistor lookup table, generated with --num-temps=50 and trimmed in lower temperature ranges.
// You may be able to improve the accuracy of this table in various ways.
//   1. Measure the actual resistance of the resistor. It's "nominally" 4.7K, but that's ± 5%.
//   2. Measure the actual beta of your thermistor:http://reprap.org/wiki/MeasuringThermistorBeta
//   3. Generate more table entries than you need, then trim down the ones in uninteresting ranges. (done)
// In either case you'll have to regenerate this table, which requires python, which is difficult to install on windows.
// Since you'll have to do some testing to determine the correct temperature for your application anyway, you
// may decide that the effort isn't worth it. Who cares if it's reporting the "right" temperature as long as it's
// keeping the temperature steady enough to print, right?
// ./createTemperatureLookup.py --r0=100000 --t0=25 --r1=0 --r2=4700 --beta=4066 --max-adc=1023
// r0: 100000
// t0: 25
// r1: 0
// r2: 4700
// beta: 4066
// max adc: 1023
#define NUMTEMPS 20
// {ADC, temp*4 }, // temp
const uint16_t PROGMEM temptable[NUMTABLES][NUMTEMPS][2] = {
{
   {1, 3364}, // 841.027617469 C
   {21, 1329}, // 332.486789769 C
   {41, 1104}, // 276.102666373 C
   {61, 987}, // 246.756060004 C
   {81, 909}, // 227.268080588 C
   {101, 851}, // 212.78847342 C
   {121, 805}, // 201.30176775 C
   {141, 767}, // 191.787692666 C
   {161, 734}, // 183.662212795 C
   {181, 706}, // 176.561442671 C
   {201, 680}, // 170.244089549 C
   {221, 658}, // 164.542298163 C
   {241, 637}, // 159.33475843 C
   {321, 567}, // 141.921298995 C
   {381, 524}, // 131.166509425 C
   {581, 406}, // 101.561865389 C
   {781, 291}, // 72.9710018071 C
   {881, 219}, // 54.8051659223 C
   {981, 93}, // 23.4825243529 C
   {1010, 1} // 0.498606463441 C
}
};

Wurstnase, 2020-03-04:
I think I have an idea.

You named the heater in your board config `HEATER_HOTEND` and try to read the temperature with `M105`, right?

You have 2 options. Read the temperature with `M105 P0` or rename the `HEATER_HOTEND` to `HEATER_EXTRUDER`. Default procedure is, that the code is searching for `HEATER_EXTRUDER` but this is not defined in your case.

azant, 2020-03-04:
Yes! You’re correct. I tried using M105 & M105 T0/T1. I will try tonight using M105 P0 and if it doesn’t work then I’ll re-name it. Thank you :)

azant, 2020-03-05:
> I think I have an idea.
>
> You named the heater in your board config `HEATER_HOTEND` and try to read the temperature with `M105`, right?
>
> You have 2 options. Read the temperature with `M105 P0` or [...]

i used M105 P0 and it did read the Temp. but re-named it to what you have suggested Pronterface still reads 0. not sure if thats normal. Thanks

azant, 2020-03-05:
I think i figured it out. Thanks to you! i re-named it to just extruder and sent M105 P0, i also used "gettemp" which reported Hotend 0/0. then temperature started to show up on pronterface.

Wurstnase, 2020-03-05:
When you're sending a M105 you should see also the temperature now. Did you
test this also?

azant, 2020-03-05:
> When you're sending a M105 you should see also the temperature now. Did you test this also?

Yes! that worked! it is reporting temp. readings. 

working on PID Overshoot. even though i changed the Temp Hysis. from 10 to 2 it still overshoot 10 degrees from setpoint.

Wurstnase, 2020-03-06:
You need to set the PID values. Overshooting is too much integration, so you need to decrease the `DEFAULT_I` in heater.h. Unfortunately there is currenlty no setting in the config files.

AnHardt, 2020-03-06:
>...  Overshooting is too much integration ...

The I-term of the PID is only one factor for the overshoot. 
The other is death-time. A change at the heater has to creep thru the heater-block until it arrives, some time later, at the sensor. The on/off frequency and error amplitude of a simple setpoint-on/off controller is completely determined by that.

Wurstnase, 2020-03-06:
Hmmm... Too much time ago I played with my printer.
First you need to decrease the P-part from PID until overshooting is minimized. Not the I-part.

Unfortunately this time the german wiki is much better than the english one. So I don't have a good link to help through.

azant, 2020-03-06:
Thanks guys! Appreciate your support. I’ll mess with it on the weekend. 

I also noticed that when I unchecked the BANG BANG BED control, my extruded doesn’t heat. I don’t see any output to heat.

closed 2020-03-10 by azant