Traumflug/Teacup_Firmware #324: Teacup freezes during print

azant, 2020-03-13:
Last night I finally had everything working and calibrated. I tried to test print a dice it was working well and froze 20 minutes later. 

I tried to cancel the job. Manual jog didn’t work, reset or restart print. Disconnected serial cable and plugged back in, I got it to work after I cycled power to all drives and mcu. I tried different 3d models to print same thing happened. 

I used slic3r and pronterface.

Thanks

Wurstnase, 2020-03-17:
Hi Azant,

did it freeze always after ~20 minutes?
Could you reduce the problem to a specific command, send from pronterface or the slic3r created gcode?
Do you need to power cycle your computer and/or the Arduino? Do you use Windows or Linux?

It would help to find the issue when you upload **_any_** file you changed for your printer. Maybe it is a good idea to upload your complete teacup folder into your own github repository. If you need help with this, please let me know.

Also I need the gcode file you which break after 20 minutes.

azant, 2020-03-17:
> Hi Azant,
>
> did it freeze always after ~20 minutes?
> Could you reduce the problem to a specific command, send from pronterface or the slic3r created gcode?
> Do you need to power cycle your computer [...]

it freezes after 30min~60mins! i tried slowing down print speed, that didnt help either.

what i found out if i send M114 it resumes printing but doesnt report print status or pause or abort. I do unplug the USB cable and re-connect and start all over, i noticed that happens for longer prints. I used windows.

i will upload my teacup folder tonight.

arcimede, 2020-03-18:
Hi 
as I have setup an Teacup Printer-Controller recently I noticed, that the printing stopped sometimes due to a communication problem. Lowering the baudrate seems to help, i use 57600baud now with Octoprint...

Good luck
Joerg

azant, 2020-03-20:
Thanks guys! @arcimede @Wurstnase. I lowered the Baud rate to 57600 and i replaced the USB cable seems to work. i ran a test and printed for 2 hours and didnt stop.

is there way to config a FAN PIN ? i did not see that on the configtool.

Thanks

Wurstnase, 2020-03-20:
Take a look into the heater module. There should be something you want to add.

azant, 2020-03-20:
> 
> 
> Take a look into the heater module. There should be something you want to add.

Thank you!

azant, 2020-03-29:
Reducing baud-rate and switched to Octopi/print works perfect haven't had any interruption .

I look at adding a fan through heater module . but it forces me with 2 pins D11 and D3. 
FYI, I'm Using atmega328p.

Wurstnase, 2020-03-29:
What forces you? The configtool?

Can you please upload your config files and thermistor file? I would like to check file size etc.

azant, 2020-03-29:
> 
> 
> What forces you? The configtool?
> 
> Can you please upload your config files and thermistor file? I would like to check file size etc.

[printer.tdog.zip](https://github.com/Traumflug/Teacup_Firmware/files/4399293/printer.tdog.zip)

the configtool, i used DIO11 for Hotend so it will only list DOI3

Wurstnase, 2020-03-29:
You could enter any pin in the config file directly. Probably the config tool will show only pwm-able pins, which isn't necessary anymore.

I will check the pin parser the next days. I'm currently waiting for a new display for my laptop.

azant, 2020-03-30:
> 
> 
> You could enter any pin in the config file directly. Probably the config tool will show only pwm-able pins, which isn't necessary anymore.
> 
> I will check the pin parser the next days. I'm currently waiting for a new display for my laptop.

I Got it! Thanks again. I added the PIN directly to config file. to have a software PWM, do i have to checkbox "FORCE software PWM" ? will that impact Hardware PWM ?
Thanks

Wurstnase, 2020-03-30:
No, normally you don't need that switch.
This switch will force a pin if set to 1 to soft-pwm, also when hardware-pwm would be available.

azant, 2020-03-30:
> 
> 
> No, normally you don't need that switch.
> This switch will force a pin if set to 1 to soft-pwm, also when hardware-pwm would be available.

Cool! Thank you again!

closed 2020-03-30 by azant