Traumflug/Teacup_Firmware #328: configtool colors all wrong in Ubuntu

phord, 2020-07-31:
I don't know if I can recreate this.  I was in Ubuntu 20.04 and the colors in configtool are all messed up.  I assumed it's because of my dark system theme, so I went to settings to change them to light.  But found Ubuntu's settings didn't have _any_ theme selected. As soon I chose one, the ConfigTool became readable again.

[image]

Maybe not reasonable to fix, but logging this in case someone else repros it later.

phord, 2020-07-31:
Which I switch tabs, the colors remain screwy.  The correct if I choose a theme in the System settings, but which I switch tabs, they go nuts again.  So, this is still broken.  :-(
[image]

Wurstnase, 2020-09-02:
Hmmm... Probably we can force the colors somewhere for wx. I think there is a switch for using/not using "system settings" or similar.

Wurstnase, 2020-09-04:
@phord , which version of wx are you using. I found some issues with mac os and 4.0x. Maybe it's solved with 4.1?