Traumflug/Teacup_Firmware #330: Build broken on Fedora 34

dwillmore, 2022-03-01:
This is likely user error, but I'm completely new to Teacup and I'm probably holding it wrong.

I'm following the [Simple Installation](https://reprap.org/wiki/Teacup_Firmware#Simple_Installation) instructions.

I cloned the repo and fired up the config tool. I selected the CNCv3 shield as the board and a Mendel as the printer--I just have a UNO with a CNC shield that I want to work with and there is no printer involved.  After saving config.h, I went to build and I get the error:

`/usr/lib/gcc/avr/10.2.0/../../../../avr/bin/ld: /home/willmore/src_new/Teacup_Firmware/build/mendel.o (symbol from plugin): in function `init':
(.text+0x0): multiple definition of `err_codes'; /home/willmore/src_new/Teacup_Firmware/build/intercom.o (symbol from plugin):(.text+0x0): first defined here
lto-wrapper: warning: Options to Xassembler do not match: -adhlns=/home/willmore/src_new/Teacup_Firmware/build/display_ssd1306.al, -adhlns=/home/willmore/src_new/Teacup_Firmware/build/delay.al, dropping all -Xassembler and -Wa options.
collect2: error: ld returned 1 exit status
RC = 1 - Build terminated
Link terminated abnormally.`

I can't find any references to err_codes in the source, so I'm not sure where it's picking that up.  Any advice would be welcome.

phord, 2022-03-08:
Hm.  `err_codes` is defined in the intercom.h file, and also in the extruder/intercom.h file.  I wonder if your model is including both of them somewhere.  

Can you try renaming one of them and see if it helps?  The actual type doesn't seem to be used explicitly anywhere, so the name of the enum is unimportant.

dwillmore, 2022-03-09:
Thank you for the hint.  I commented out one of the copies of the structure
and it builds, now.  The odd thing is that nothing seems to ever refer to
that structure in the executable parts of the code.

phord, 2022-03-09:
That's common in C.  The enumerated values are often the only part that matters. In this case, it's `ERROR_BAD_CRC`.

dwillmore, 2022-10-11:
You're correct, sorry, brain needed engaged before mouth.