Traumflug/Teacup_Firmware #334: Support fot PT100 sensor

ahmetonat, 2023-06-07:
In my quest to install Teacup firmware into an Atmega 328 so that I can resurrect an old Tiertime UP-Mini, I stumbled into another important problem. The Up Mini uses a thermocouple of type PT100. Sure, no problem; Teacup does have a PT100 in the temperature sensors section. But wait; why does Repetier host show 1835degC when I select it?

Diving into the code searching for the term PT100:
temp.c:    #ifdef TEMP_PT100
temp.c:      case TT_PT100:
temp.c:        #warning TODO: PT100 code

Oh well...

So my question is, I am willing to vounteer for the PT100 code support. Can anyone give me any hint on :
- Where to start,
- What is the circuit model for the PT00 sensor (what type of slope, what offset? (I saw instrumentation amplifiers for thermocouples used in Ramps printers. Up Mini seems to have a different circuit.)
- And possibly: Some code from another firmware that I can use as a starting point? 

Ahmet Onat

Traumflug, 2023-06-07:
Well, I can't really help you, as it's too far back since I looked into this code. My little machine simply works and works, no need to touch anything.

Regarding PT100 I vaguely remember that a first step would be to try to use the existing calibration system for this type of sensor as well. Thermistors and Thermocouples are different technology, but behave similarly, just in a vastly different parameter range.

ahmetonat, 2023-06-18:
The sensor in the Up Mini has a linear characteristic. After some attempts at creating tables with linear characteristics (I modified configtool/thermistortablefile.h and added a new function which creates a new table for linear sensor under the "thermistor" group just to test the idea), the sensor still did not run report the correct temperatures. I had no other physical sensor to read the correct head temperature so it was not possible to do any sort of meaningful calibration.

Eventually I wrote a simple Arduino code that would do the following:
- After power ON, it enables the motors and
- Reports the head temp sensor ADC reading through the serial port.
- When the Y limit switch is pressed, it turns on the heater
- The extruder stepper is physically unplugged.
- When the heater is sufficiety hot and the filament is pushed in by hand, it comes out of the nozzle.

By pushing PLA through the nozzle, it was seen that it melts nicely at around ADC=325, which corresponds to about 160 degC.  Also at room temp (25 deg C), the sensor reads 50. By using these two data points and since the sensor has a linear characteristic, the math is then quite simple, the ADC reading must be divided in half to get the temp value.  (Since Teacup uses 4*T, actually ADC value must be *multiplied* by 2.)

I simply went into temp.c and modified 
`static uint16_t temp_table_lookup(uint16_t temp, uint8_t sensor)`
function so that it simply returns the analog read value multiplied by 2 as the current temperature.

This works quite nicely.

At first I wanted to add a new selection to the temperature sensor window of the GUI and make the necessary additions, but the code is quite difficult to modify around those parts and the simple hack described above worked nicely. Sorry for machetteing my way through the code and not doing a nice addition but since this printer hardware is essentially obsolete for many years now, I assume there is not much interest. If someone needs, I can try to implement a proper modification which can then be pulled in.

closed 2023-06-25 by ahmetonat