Traumflug/Teacup_Firmware #335: Homing direction cannot be set

ahmetonat, 2023-06-18:
I have an Up Mini printer that I am trying to retrofit with an Arduino Nano and Teacup firmware.

It has a belt driven Z axis which has a limit switch which is triggered when the platform is at the lowest position (maximum Z position). So the Homing Order  in <Printer> <Mechanical> should be something like:
`Step 1: x_negative
Step 2: y_negative
Step 3: z_positive
Step 4: none`

However, this has no effect, the printer always tries to go to minimum Z for homing, regardless.

This setting is configured in the file
`printer.UpMini.v0.1.h ` (or whatever your printer name is)
However, there is no `#define` in front of that definition. It looks like this:
`/** \def DEFINE_HOMING
  Order (and number) of homing movements. Up to 4 homing steps are allowed.
  If you don't need even one axis just DEFINE_HOMING(none).
*/
DEFINE_HOMING(x_negative, y_negative, z_positive)`

If I add a `#define`  in front of it by hand (and do a manual build: `make -f Makefile-AVR`) the build fails because the same value is also defined in `config_wrapper.h`.

Any ideas?
 
Ahmet

Wurstnase, 2023-06-18:
Maybe your endstop is triggered?

ahmetonat, 2023-06-18:
I think not, because when I home the Z axis the platform starts to move up (that decreases the distance to the head, so it is in the negative direction). When I manually trigger the switch, it stops the motion. From that point on, it works normally. Z has beem homed correctly etc.

Only the homing direction is wrong.
Here are the screenshots of the related parts:

[image]

[image]

ahmetonat, 2023-06-25:
After modifying the settings in the pinouts page, I was able to obtain a correct homing and operation directions. Everything works correctly now.

The description of the "Invert Maximum"  tickbox can be written in. It is not clear whether it means the logic level of the limit switch at touch, or something else. I have 3 limit switches and they are all set the same. In my case, I had to de-select "Invert Maximum **only** on the Z axis for it to work. So it probably means something other than the logic state.

Also, since my Z limit switch is at the bottom of the printer (farthest from the nozzle), it is important to set the "Max-Z" in Teacup -> Printer -> Mechanical tab correctly, and this number **must** be the same as the dimensions of the printer in the G-Code sender (repetier host etc.). Then it will work correctly. The current setting screenshot included here for reference. Note only "Invert Maximum" for Z axis is unchecked.

[image]

closed 2023-06-25 by ahmetonat