Traumflug/Teacup_Firmware #338: My awesome journey ends here

Traumflug, 2023-08-20:
Folks, Github is about to enforce 2FA (Two Factor Authentification).  That's IMHO a pretty pointless burden.  Quality control in Open Source isn't established by some door lock, but by eyeballs looking at the code.  Git commit IDs are enough of a verification, same ID gives always the exactly same code (and even the very same commit message).  And yes, Git has also a mechanism to sign tags and commits with a GPG signature.  That's as safe as public code can be.

I'm not going to mess with my phone (only way to give a second factor) when doing development, so I'll lose control over this repository in a couple of days.  For this reason, I'm about to archive this repository. All my Github activities requiring 2FA will chease for the same reason.

Should I happen to pick up development on this project again, it'll appear on GitLab, or some other appropriate public Git server. Archived repositories stay readable, and I also have my local copy.

That said, this project has a (pretty large) number of collaborators.  Should one of you, @amsler, @Cyberwizzard, @dcousens, @drf5n, @jgrjgr, @konoppo, @Laczen, @mattgilbertnet, @nesqi, @NickE37, @nikkiverre, @phord, @RaD, @rkonklex, @RobertKuhlmann, @rollingdice, @ryannining, @stecor7001, @techboycr, @triffid, @Wurstnase, @youssefaly97, @zuzuf feel like continueing with this project, please speak up now.  I'll happily transfer repository ownership over to you.

At this point, let me say a big **Thank You** to all contributors to Teacup Firmware.  It's still an excellent piece of software, a masterpiece of low-level embedded software, and it's _you_ who made this possible.  Special thanks to @triffid, of course, who started this project and gave it its structure.  I'm still stunned what one can do with just sophisticated Makefiles, even with a user friendly GUI.  And I'm still stunned how well this G-code decoder works.  It can keep up with characters coming in at 115200 baud, on a CPU as low level as an ATmega.  Awesome work!

Traumflug

triffid, 2023-08-20:
If you do end up moving this to gitlab or whatever, put a link in the README before you archive so folk can find it easily.

It's been an absolute pleasure to watch you take my dinky little fun project and manage its growth as much as you have, seems like it was an excellent idea for me to hand over the reins when it became clear that you were more interested in developing it than I was!

For other readers who don't know the background of Teacup, the impetus to initially create this project was the 2009-era reprap community telling me that one atmega wasn't big/fast enough to handle everything by itself (which technically is true-ish if you're using float math in interrupt context like the firmwares available at the time, and want to enable/disable steppers individually) and I wanted to create & explore a different structural paradigm that _would_ be able to do all the things - which means no float math, no blocking code _anywhere_, a heavy focus on cpu clock efficiency, and may as well throw in some compile-time structure building with macro and makefile shenanigans for fun.

I may have laid a foundation, but you've done the bulk of building upon it and managing others' contributions - thank you!

PS: https://imgur.com/a/E0jb7 shows the hardware that Teacup was initially developed on if anyone's curious

Traumflug, 2023-08-20:
Thanks. There is no GitLab project yet, nor one planned in the near future, so I fear I have no link I could put into the README. My nice little machine works just fine, no inducement to change the code.